ros-ukraine/leobot #2: Create Logo

AndriyPt, 2017-11-20:
Create logo of the robot.
Robot name is LeoBot so it should be something with Lion :)

SystemDiagnosticss, 2017-11-30:
Одним словом не знаю як грузити сюда .psd файли, тому даю ссилку на файлообмінник http://dropmefiles.com/1lg6q - psd-файл, http://ibb.co/ev6e4b - картинка в jpeg. Шрифт - https://fontlibrary.org/ru/font/glacial-indifference, емблема - з голови.

AndriyPt, 2017-11-30:
Looks amazing :+1: 

1. Assign yourself to this issue as it is described here  https://help.github.com/articles/assigning-issues-and-pull-requests-to-other-github-users/
2. Now please create branch which starts from F#2_<your_name>. You may use UI for that https://help.github.com/articles/creating-and-deleting-branches-within-your-repository/
3. Commit your PSD and JPEG files into design folder here https://github.com/ros-ukraine/leobot/tree/kinetic-devel/docs/design
4. Create Pull Request from your branch as described here https://help.github.com/articles/creating-a-pull-request/ 
5. Put in description of Pull Request words: Closes #2 in order to automatically close this issue using this GitHub feature https://help.github.com/articles/closing-issues-using-keywords/

PR #14 (2017-12-09): Logo
Closes #2
closed 2017-12-09 by AndriyPt