ros-ukraine/leobot #3: Create simple look of robot

AndriyPt, 2017-11-20:
It should be wheeled robot
It will contain camera, speakers and microphone.
Also there should be monitor or any other device to visualise remote person.

MaxZhaloba, 2017-11-30:
Suggestion of design from SystemDi: https://ibb.co/iNhwZb

LyubomyrD, 2017-11-30:
We can increase the mobility of the robot by adding more advanced wheelbase (see the paper attached to this message). The idea from the article entirely correlates with the idea if changing the hight of the robots screen/face.

The model of the wheelbase is given in the paper which may help us when reproducing it in CAD model.

What do You think?

[10.1142@S1793962317500155.pdf](https://github.com/ros-ukraine/leobot/files/1519201/10.1142.S1793962317500155.pdf)

AndriyPt, 2017-11-30:
@LyubomyrD very interesting approach. But maybe it is worth to start from simpler version of the wheels and after that increase complexity. Since it might be too complicated to simulate and control such robot on phase #1. Maybe it is something which we can do during the next iteration of the hardware :)

SystemDiagnosticss, 2017-11-30:
Любомир, то реально круто!!

LyubomyrD, 2017-11-30:
OK. Let's move on faster with the simple version.

abench, 2017-12-02:
Proposal of @LyubomyrD is very good. But I agree with @AndriyPt that it is very complicated for the first project. Design of @SystemDiagnosticss is simple enough for implementation. But the stereo camera is usually absent on tablets. Should we use stereovision for the first project?

abench, 2017-12-02:
I have following simple proposal with a stereo camera.
[image]

MaxZhaloba, 2017-12-02:
Mecanum Y-shaped wheels look really cool. But I'm not sure how affordable and realistic they are for us. Another option could be to stick to simpler Y-wheels. But I agree, that we need to begin with something simpler.

@abench Your sketch looks good to me, but I have one question though. Why do you prefer 3 wheels base over 4 wheels? Do you think that 3 wheels will add stability to the platform or minimise shaking? Maybe that's the case, just asking.

SystemDiagnosticss, 2017-12-02:
Я звичайно обома руками за простоту. Але думаю ми всі трохи прибідняємся.  Варто лише глянути на команду, яка зібралась. Тут є люди які працюють на фул тайм і програмістами і електроньщиками, дехто навіть на посаді сеньйора. З нами також доктор наук в цій області, а тім-лід в нас, взагалі, з компанії, яка  зробила найкращу робо-руку НА СВІТІ ... і цей список можна продовжувати.  
Просто якшо ми зробимо робота який нічим не буде відрізнятися від сотні інших, то для кого ми будемо цікаві? В мене вдома стоїть 4 штуки роботів, якими не можна зацікавити жодного нормального інвестора .. не велика то радість скажу вам відверто.

abench, 2017-12-02:
@MaxZhaloba, 3 wheels were selected for simplicity and stability in mind. But they also can minimise shaking, as You noticed. Unexpected shaking for stereo is very annoying from my experience of seeing 360 videos taken from a drone.

AndriyPt, 2017-12-03:
@SystemDiagnosticss і у нас на підприємстві намагаємося йти від простого до складно :) Якщо нам вдасться швидко зробити простого робота, то не бачу проблем зробити складнішого, але якщо ми зразу візьмемося за складного і будемо його жувати довго і нудно, то всім від того не буде легше.
У таких проектах (де багато зайнятих людей, які не можуть приділяти стабільну кількість часу кожного дня) "краще менші кроки, але частіше, ніж великі, але рідше", то з досвіду ;)

AndriyPt, 2017-12-03:
@abench I like the design which you have proposed :+1:  It is simple enough to be implemented using basic materials and use differential drive rather than a little bit more complicated mecanum wheel (which we definitely might use on the later stages of the project).
If someone has objections please write down here your concerns else we can move on with this design :)

SystemDiagnosticss, 2017-12-04:
@AndriyPt я за простоту обома руками. Як ти скажеш так і буде. Лишень повторюсь, про ситуацію з деталями. Простий приклад: знайомий захотів зібрати собі гоночний коптер . Нормальний гоночний коптер десь вийде 1000уе. Ну він зробив за 500, бо вирішив трохи зекономити. Рівно через 3 тижні він зрозумів, що цього коптера йому недостатньо. Він вже побавився і хоче чогось реального.  Ну і в результаті купив нову раму, нові мотори, новий контроллер польотів і тд. В результаті витрав на 2 коптера замість одного. Тобто я з цієі сторони маю на увазі, шо треба все добре обдумати...

AndriyPt, 2017-12-04:
@SystemDiagnosticss ключовою різницею є те, що ми нічого не купуємо. Наразі ми будуємо симуляцію робота. Думаю, що до того моменту, коли ми будемо купувати деталі пройде 2-3 ітерації вигляду і він може бути кардинально інакшим, але наразі варто почати із найпростішої CAD моделі, яку можна буде удосконалювати, коли ми побачимо проблеми (а ми їх неодмінно знайдемо).

SystemDiagnosticss, 2017-12-04:
@AndriyPt в такому разі обома руками за!

chexov, 2017-12-04:
у вас є якийсь слак чи інший чат? хочу допомогти

AndriyPt, 2017-12-04:
Привіт @chexov :) 
Ми використовуємо GtiHub. Беріть довільну задачу до якої ніхто не причеплений. 
Якщо якісь питання, пишіть у задачі.
Анонси телеконференцій і певні обговорення можуть ще бути у Facebook групі https://www.facebook.com/groups/279366215828155/  (там і користуємося приватними чатами, якщо треба).
Наразі потреби у інтерактивному груповому чаті не було.

LyubomyrD, 2017-12-06:
Я б всетаки пропонував використовувати чотириколісну платформу з колесами меканум. Три колеса дадуть нам статичну стійкість, звісно, але додатковим колесом ми б зробили робота ще бульш ударостійким.
Також колеса меканут дозволять підвищити маневреність робота., то ж я б від них не відмовлявся попри більшу ціну. В музей може скластись ситуація, коли неголономному роботу (non-holonomic) просто не буде місця щоб повернути не розбивши чогось.
Платформу я б пропонував обрати квадратну. Розмір і вагу платформи можна спробувати визначити залежно від довжини та ваги "шиї робота".
m1 - маса шиї
m2 - маса основи
L1 - довжина шиї
L2 - сторона основи.
g - сила земного тяжіння
a - прискорення яке надасить роботу ніжне плече старшокласника прикладене до середини шиї робота
Тоді
m2/2*g*L2/2=m1*a*L1/2
(сподіваюсь, я не зробив дурної помилки з формулі)

Стосовно датчиків, то я б попонував спостерігати за роботом в 360 градусах. Такі лідари точно є. Питання ціни.

Підтримую ідею створення кількох "площин" в яких ми спостерігаємо за роботом: основа, голова і можливо для підстраховки кілька по середині шиї.

Тобто я б пропонував сумістити усі ідеї які тут висловлені:

1) Колеса Меканум
2) Квадратна основа
3) Рухома шия
4) Камера зверху (і планшет)
5) Кілька площин спостереження за перешкодами. Можливо дві лідари?
В Autonomous Robots (Siegart) стверджують, що лідар можна земулювати 56 ультразвуковими датчиками. Трохи складно, як на мене.
6)  Датчики(можливо бампер) для аварійних ситуацій. 

Що ви про це думаєте?
 
P.S. Я, на жаль, зовсім невмію малювати. Перепрошую.

SystemDiagnosticss, 2017-12-06:
Підтримую Любомира, щодо 4 колісноі основи.
Також ще раз хотів би винести питання на рахунок можливості руху шиі робота вверх- вниз. Тобто в разі чогось, щоб робот міг пригнутися. Ну і в дуже багатьох ситуаціях я думаю виникне питання сходинки ... Ну тобто щоб будь який бордюр не став кінцевою точкою нашого робота ....

SystemDiagnosticss, 2017-12-06:
І ото питання сходинки фактично не реалізовано в більшості роботів телеприсутності. Якшо ми подолаєм це питання, то зможемо легко видіоятися з великоі маси роботів телеприсутності, що представлені на ринку

SystemDiagnosticss, 2017-12-06:
@LyubomyrD,  можна впринципі спробувати поставити лідар ще один на основу де буде планшет кріпитися. Або використати 3д сканер ... Хоча для навігаціі як розумію дані з 3д датчика все одну будуть перетворюватися в 2д дані ...

LyubomyrD, 2017-12-06:
@SystemDiagnosticss треба порівняти по цінах лідари і 3D сканери. 
@AndriyPt, ти працював з 3D сканером?

Я б ще хотів звернути вашу увагу на відео яке запостив @AndriyPt  в групі Facebook. Там на 02:25 є робот з цікавим розташуванням коліс. Може переймемо в них досвід?
https://prajankya.me/articles/lidar-robot

nemo8, 2017-12-06:
Краще коли платформа 4-на, але сама основа платформи кругла - при повороті на місці не зачіпатиме нічого навколо.

LyubomyrD, 2017-12-06:
@nemo8 я не вірю, що нас з гусеницями впустять хоч в якийсь музей.

nemo8, 2017-12-06:
гусениці можуть бути резинові як і колеса.

MaxZhaloba, 2017-12-06:
@LyubomyrD Підтримую вцілому, але на першу емульовану версію я би все-таки робив щось просте, навіть тривіальне. Тобто:

1. Стандартні круглі колеса (диференціал)
2. Квадратна основа
3. Статична шия
4. Камера зверху (із або без планшета)
5. У процесі обговорення: спостереження за перешкодами
6. Датчики(можливо бампер) для аварійних ситуацій

chexov, 2017-12-12:
Як на рахунок взяти розміри і платформу "як румба"?

AndriyPt, 2017-12-14:
@chexov думаю воно десь так і має вийти :)

LyubomyrD, 2017-12-14:
Платформа "як румба" буде дуже важка або мати малу "шию" або дуже нестійка.

closed 2018-04-01 by AndriyPt