ros-ukraine/leobot #6: Create CAD model of robot

AndriyPt, 2017-11-20:
Create CAD model of robot in any available application and export it to STL or DAE formats.

chexov, 2017-12-12:
So I am starting to think about base CAD model based on drawings and ideas from other issues...
Any ideas are welcome here

AndriyPt, 2017-12-14:
@chexov unfortunately we are just starting to work on the project so people not always assign themselves to the issues to indicate that they are working on it. As far as I know @abench is working on this issue.
@abench could you please assign yourself to issue if you are working on it :)
@chexov please have a look on #15 or #16 which I have recently created. And please assign yourself to issue on which you are working ;)

abench, 2017-12-15:
@AndriyPt I assigned himself to this task

abench, 2017-12-28:
First version of robot base. All are welcome to add suggestions and comments
[image]

chexov, 2017-12-28:
@abench its fine for first draft. But.
What are your plans to share source files? Have you consider using Onshape or Fusion360 so we can collaborate on design?

I would remove all the fillets at first iterations and make it easier to manufacture, create 1:2 or 1:10 models using comodity components.
Please hold in mind that this is real object. And real components will be used in it. For example if platform will use nema17 steppers there should be at least mount for such motor in mind.
Please consider putting "object for scale" nearby so team can get feeling what size is this model.

AndriyPt, 2017-12-29:
@chexov this CAD model was created using OnShape. 
Regarding manufacturing it is something outside of scope of this task since we would like to produce a little bit advance platform with mecanum wheels. This one is for simulation only.
Again "object for scale" very good idea :+1: 
@abench I will definitely recommend to have platform rotationally symmetrical to decrease chances of complicated collisions. Like in this case "tail" could be trapped under low furniture :) But in general design looks cool :+1:

closed 2018-04-01 by AndriyPt