ros-ukraine/leobot #15: Create HTML and CSS for design

AndriyPt, 2017-12-14:
Create HTML, CSS for design which was created in #4 
Please place it in the folder ./docs/design/management 
You can use design link from #4 comments

AndriyPt, 2017-12-17:
Design can be found here https://github.com/ros-ukraine/leobot/blob/kinetic-devel/docs/design

YarynaD, 2018-01-09:
Hello! 

@AndriyPt, I have pushed html and css files to the branch F#15_create_html_css.

@ros-ukraine/leobot please, have a look. If everything is ok, I'll create a pull request.

YarynaD, 2018-01-09:
Website layout was tested for large, medium (tab) and small (smartphones) screens. I tested it in Windows and in Linux. If someone can also check it on Mac, that would be great!

And I hope nobody uses IE anymore :)

AndriyPt, 2018-01-09:
Hello @YarynaD 
Please create Pull Request to get code review. This is main entity of GitHub which allows effective code review :)

YarynaD, 2018-01-09:
I've created the pull request. All checks have passed. Now a review from someone with write access is required :)

AndriyPt, 2018-01-19:
Completed by PR https://github.com/ros-ukraine/leobot/pull/23

YarynaD, 2018-01-22:
Hello! 
@MaxZhaloba, I've seen your photos of the Leobot website on FB. It seems that design is broken. I didn't experience this problem while testing the page.  Please let me know which browser you use on your phone and what is it's screen resolution. I want to fix media.

MaxZhaloba, 2018-01-22:
Hi, @YarynaD 

Cool :) Thanks for quick reaction. I haven't noticed any issues though :)

I opened it on Safari under iOS 11.2.2. I cannot find any version number associated particularly with the browser. The screen size corresponds to iPhone 5s.

Here are the details I received from http://whatsmyuseragent.com/

User Agent:

```Mozilla/5.0 (iPhone; CPU iPhone OS 11_2_2 like Mac OS X) AppleWebKit/604.4.7 (KHTML, like Gecko) Version/11.0 Mobile/15C202 Safari/604.1```

Browser Information:
```
JavaScript Enabled:	Yes
Cookies Enabled:	Yes
Device Pixel Ratio:	2
Screen Resolution:	320px x 568px
Browser Window Size:	320 px x 454 px
```

Please let me know if you'd need anything else.

YarynaD, 2018-01-23:
@MaxZhaloba, thank you very much for your fast reply! 

The information is very detailed, thank you! 
It will be very useful for me, trying to fix the design.

AndriyPt, 2018-02-09:
@YarynaD have you already fixed this design issue? Can I close this ticket?

MaxZhaloba, 2018-02-09:
@YarynaD, in case of any further updates, please commit them to leobot_webserver/web_content folder in F16 branch. Thanks!

YarynaD, 2018-02-09:
@AndriyPt  I fixed those design issues. If any other problems with design will show up, I'll gladly fix them.

@MaxZhaloba sure!

closed 2018-02-09 by YarynaD