ros-ukraine/leobot #16: Create web package

AndriyPt, 2017-12-14:
Create ROS package which will:

- start web server which can run nodejs 
- renders page with text "Hello World"

MaxZhaloba, 2017-12-30:
I've investigated the existing packages and some manuals. So far it seems that the new package will reside at `/leobot_web_server` in repository which will map to `~/workspace/leobot/base/src/leobot/leobot_web_server` in docker container.

Findings and future steps:
1. The package requires `package.xml` and `CMakeLists.txt` in its folder.
2. Also, we probably need a launch file at `/leobot_web_server/launch/web_server.launch` which will run the web server in standalone mode. This launch file will be triggered by the main lauch file when leobot is loaded as a whole.
3. This package will contain the JS server code and static web content in future.

Feel free to share your suggestions.

MaxZhaloba, 2017-12-30:
Example of generic ROS package:
https://github.com/ethz-asl/ros_best_practices/tree/master/ros_package_template

MaxZhaloba, 2017-12-30:
Just adding a few more links to primary documentation in case somebody would work on a similar task.

Description of project structure (i.e. `~/workspace/leobot/base` and `~/workspace/leobot/base_deps` in this repository):
http://wiki.ros.org/catkin/workspaces

Documentation on packages:
http://wiki.ros.org/Packages
Typical package structure:
http://wiki.ros.org/ROS/Tutorials/CreatingPackage
It's easier to generate a new package using `catkin_create_pkg` utility provided with ROS.

Documentation on launch-files:
1. http://wiki.ros.org/roslaunch
2. http://wiki.ros.org/roslaunch/XML

MaxZhaloba, 2018-01-09:
I've investigated the ROS features to run 3rd-party executables from launch files. It seems that a simple instruction `<node type="node [args]">` should do the job. But before that, Node JS must be installed in the container. My plan is to include the `apt-get install nodejs` instruction to the `/Dockerfile` and rebuild the container (still need to find out how to do this).

If anyone would see any major pitfalls with this approach please let me know.

AndriyPt, 2018-01-10:
@MaxZhaloba ROS has system called rosdep http://wiki.ros.org/ROS/Tutorials/rosdep which is used to load system dependencies specified in package.xml.
I have looked up packages for you in their mapping file https://github.com/ros/rosdistro/search?utf8=%E2%9C%93&q=nodejs&type= and https://github.com/ros/rosdistro/search?utf8=%E2%9C%93&q=npm&type= 
This mapping file is used to map name from package.xml to apt-get package name or pip package name in case of python.
I will suggest you to read more about rosdep and add appropriate dependencies to package.xml. Docker container will install these packages automatically since it is using rosdep already :)

MaxZhaloba, 2018-01-11:
@AndriyPt Great! That's very useful for me.

I've proceeded with rosdep and added a dependency for `<depend>nodejs</depend>` to the `package.xml`. I've also updated the launch file with such dummy invocation of Node JS:

```
<node pkg="web_server" type="nodejs" args="-h" name="web_server" output="screen"/>
```

I've hit an issue that Node JS cannot be found when I'm running `roslaunch web_server.launch`. 

```
└──╼ roslaunch  web_server.launch
... logging to /home/user/.ros/log/8aa1b702-f725-11e7-8d9b-0242ac110002/roslaunch-a8103f763894-23086.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://a8103f763894:34041/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.12

NODES
  /
    web_server (web_server/nodejs)

auto-starting new master
process[master]: started with pid [23097]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 8aa1b702-f725-11e7-8d9b-0242ac110002
process[rosout-1]: started with pid [23110]
started core service [/rosout]
ERROR: cannot launch node of type [web_server/nodejs]: web_server
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/user/workspace/leobot/base/src
ROS path [2]=/home/user/workspace/leobot/base_deps/src
ROS path [3]=/opt/ros/kinetic/share
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
I've tried to troubleshoot this issue with `rosdep install`:

```
└──╼ rosdep install leobot_web_server
dpkg-query: no packages found matching nodejs
dpkg-query: no packages found matching nodejs
executing command [sudo -H apt-get install nodejs]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package nodejs
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install nodejs] failed
```

So, for some reason, "nodejs" dependency cannot be found in rosdep as such.

It seems that I'm missing some very basic knowledge or concepts of rosdep or packages. I searched for usage examples of system packages but haven't found anything similar. Can you please suggest some manuals or sources of information to read?

MaxZhaloba, 2018-01-17:
Thanks, @AndriyPt  :)

I'm just transferring our conversation from FB so that everyone could read it.

I understood the approach with rosdep, but I have several questions:

1. а) So far I thought that the project will be launched with one command. Just from how I see it, the build system should download the necessary dependencies on its own and launch the node. With current approach, if any other developer downloads the project, he will need to run the `rosdep install` and then only the launch file. Is this a typical approach? If so, then it's necessary to add `rosdep install` to documentation so that others could build the project.

Answer from @AndriyPt: We will run `rosdep install` after each update of code and we need to specify it in readme.md.

1. b) However a developer can install the necessary dependencies manually, it's not clear what to do on build servers, because they won't install any dependencies without such command and the builds will fail. If I'm not mistaken, you mentioned that such thing is configured there, but I don't know where and if this command is different from local triggering of rosdep then.

Answer: Build server is run on clean Docker image each time with minimal ROS. All necessary dependencies are installed using rosdep which reads the necessary packages from package.xml. You can do whatever you need locally, but in the end everything should be installed automatically.

2. I've tried to run rosdep from the branch F_nodejs_example with the command you provided. I received at output that rosdep cannot find packages.

```
┌─[user@a8103f763894]─[~/workspace/leobot/base]
└──╼ rosdep install --from-paths src --ignore-src --rosdistro kinetic -y
dpkg-query: no packages found matching npm
dpkg-query: no packages found matching nodejs
dpkg-query: no packages found matching npm
dpkg-query: no packages found matching nodejs
executing command [sudo -H apt-get install -y npm]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package npm
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y npm] failed
```

I've uploaded my version of code to my branch. It gives the same result. I cannot find what I'm doing wrong.

P.S. My code is in the branch F#16_web_server. Can you please have a look into this issue with "no packages found" error and give me some hints? Thanks so much!

AndriyPt, 2018-01-19:
@MaxZhaloba try to run ```sudo apt-get update``` and run `rosdep` command again :)

MaxZhaloba, 2018-01-21:
It worked! Thanks 🙂

MaxZhaloba, 2018-01-21:
This issue required adding shebang and updaing attributes for the main JS file with `chmod 755 ./web_server.js` to make it callable from `web_server.launch`.

MaxZhaloba, 2018-01-21:
Just an update on current status. I've configured a simple static web server for the content in folder `leobot/docs/design/management/Leobot_html` The latest code is found in the branch https://github.com/ros-ukraine/leobot/tree/F%2316_web_server

Further tasks:
* Stop the server when user interrupts the roslaunch with Ctrl+C. Currently the server keeps running and using the port.
* Expose the port from docker container to host.
* Configure parameters for the launch file such as server port and default web folder to serve.

MaxZhaloba, 2018-01-21:
Also, it will be necessary to automate the module initial configuration. There's a bunch of commands that need to be triggered to prepare the web server ([Starting the web server](https://github.com/ros-ukraine/leobot/tree/F%2316_web_server#starting-the-web-server)). @AndriyPt, what's the typical approach for module initialisation in such cases? How does it sound if I create an `init.sh` script in module folder for this? Should we then call it from `Dockerfile`?

MaxZhaloba, 2018-01-21:
I've updated the readme.md commands to enable port forwarding. This requires re-building the docker container to be able to open the web page from host Ubuntu OS or LAN (item #7 `docker run ...` in readme file).

AndriyPt, 2018-01-22:
@MaxZhaloba 

1. web server should contain web site in the package's subfolder called `site` or similar.
2. `init.sh` is not needed. People who see that there are some missing dependencies will run command `depkinetic` in `~/workspace/leobot/base/src` and all missing packages will be installed.
3. Regarding `npm` items installation process should be incorporated into CMakeList.txt file to run during each build. For now I could not find fastest and smoothest solution. Please check https://cmake.org/cmake/help/v3.0/command/add_custom_command.html 
4. Docker container is not build it is created from Docker image as copy of layers. Docker image is build. Docker container should be deleted and created to have new ports exposed. Also please use 8080 by default since 80 might be taken by other applications.

MaxZhaloba, 2018-02-08:
@AndriyPt, I've implemented almost all items we've discussed and pushed the code to F#16_web_server.

The remaining issues are:
1. Reading of `port` parameter in web_server.js file. Currently 8080 is a hard-coded value.
2. Web server continues to run after I interrupt the execution with Ctrl+C. This prevents the server from starting for the second time. So far I bypass this issue with `sudo pkill nodejs`.

AndriyPt, 2018-02-09:
Please create PR for review :)

PR #44 (2018-02-10): F#16 web server
Almost closes #16.
closed 2018-02-10 by AndriyPt