ros-ukraine/leobot #19: Questions re: installation of Ubuntu+ROS on a portable drive

MaxZhaloba, 2017-12-29:
It appears that my installation of Ubuntu in VMWare is working too slow. I'm going to try to install it on a USB stick using these recommendations and will get back with results.

1. Short version: https://askubuntu.com/questions/955258/how-do-i-do-a-full-installation-of-linux-onto-a-usb-flash-drive
2. https://askubuntu.com/questions/16988/how-do-i-install-ubuntu-to-a-usb-key-without-using-startup-disk-creator

MaxZhaloba, 2017-12-30:
I configured VirtualBox VM with Ubuntu installation ISO image and flash drive (just to keep the Windows usable during installation) but Ubuntu requires at least 8.6 GB free space drive. At the moment I don't have any devices with such capacity, so I'm falling back to installation to desktop SSD.

LyubomyrD, 2017-12-30:
Here are some links that might be useful for installation process

http://www.ubuntubuzz.com/2016/03/how-to-install-ubuntu-1604-xenial-xerus.html

https://ubuntuforums.org/showthread.php?t=2336099

http://bigonotation.altervista.org/en/2016/10/20/how-to-install-ubuntu-16-10-in-external-hdd/

Ihoreg, 2018-01-17:
I have just installed Ubuntu 16.04 to my Transcend 1T USB HDD connected to HB EliteBook.
I used the next recommendation:
http://www.ubuntubuzz.com/2016/03/how-to-install-ubuntu-1604-xenial-xerus.html
https://vlsu.blogspot.com/2012/04/ubuntu-usb.html
However, I bumped into one unexpected problem. The matter was that I wanted to have the 1st FAT32 big partition so long as I installed there the virtual machine (VMWare). And that caused a problem with Ubuntu bootloader (Grub Rescue message "Unknown file system").
I removed problem only when I reduced the 1st partition to 60 GB and 2nd (Ext4) partition to 40GB, so long as HB EliteBook's BIOS wasn't allow starting the bootloader from partitions upper 100 or maybe 125GB.

MaxZhaloba, 2018-01-22:
Here’s my experience with portable drives.

I’ve finally bought a USB stick after doing a comparison of the models available on our market and using benchmarks from http://usbflashspeed.com

This one turned out to have the best speed for moderate price (389 + 50 UAH for drlivery): SanDisk Cruzer Fit Ultra 32GB USB 3.0 (SDCZ43-032G-GAM46) https://rozetka.com.ua/a/p12306050

I’ve tested the r/w speeds and they’re quite impressing:

USB 2.0 - up to 28/15 MB per second (r/w):
http://usbflashspeed.com/285673

USB 3.0 - up to 150/55 MB per second:
http://usbflashspeed.com/285790

I’ve installed Ubuntu on first 14.3 GB  (BIOS loader - no UEFI) leaving the rest for FAT32. It took me 2 mins to boot from BIOS to desktop on my home PC and 33 seconds at work. Both are USB 2.0.

After downloading the docker image there remained only 500 MB on system volume, so I had to extend it to 18.6 GB. MiniTool Partition Wizard damaged the loader for some reason, so it’s better to allocate enough space in advance.

I’ve configured the project and docker running on USB 2 and as far as responsiveness is concerned it seems to work just like a normal HDD installation. I haven’t noticed any system glitches so far.

Some things such as docker image downloading, unpacking and git branch checkout take more time, but within acceptable limits. PyCharm works well too.