ros-ukraine/leobot #28: Stream video from camera to web browser

AndriyPt, 2018-01-25:
Create proof of concept solution for video streaming from one of the robot's camera to web browser.

SystemDiagnosticss, 2018-02-01:
@YarynaD Проблема з CSS. Не вдається забрати полоску що створює iframe з лівого боку 
Відповідно на верстці є 2 полоски (одна по середині, одна зліва)

YarynaD, 2018-02-03:
@SystemDiagnosticss, I'm sorry for a delay in reply. I'll check the files as soon as possible and try to figure out something in few days.

SystemDiagnosticss, 2018-02-03:
Thans, don't worry

YarynaD, 2018-02-06:
Hi, @SystemDiagnosticss. I finally loaded the files, but I don't see any line in the middle of the iframe screenshot. Have you already  fixed the problem?

SystemDiagnosticss, 2018-02-06:
@YarynaD I mean this line.
[image]
Its must be without any white line in middle or left. Its for VR set in Google cardboard.
Its must be like this
[image]

YarynaD, 2018-02-07:
Thanks. It's clear now.  

I used the <img> tag in html file. It's also supposed to be suitable for streaming. 
And in css set for container (<body> in our case) display: flex. So it will change according to screen width.
[image]

Would it be better that I upload html and css files for this screenshot? I can either send them here or commit to your branch.

SystemDiagnosticss, 2018-02-08:
YarynaD big thanks. Nice work. Please commit it to my branch

YarynaD, 2018-02-08:
@SystemDiagnosticss I've read your message in my e-mail and didn't check issue changes. As the result - I've commited changes to your branch and only now saw I should have sent them here... I hope that's ok.

closed 2018-02-16 by AndriyPt