ros-ukraine/leobot #43: Update the credits on web page

MaxZhaloba, 2018-02-08:
Currently, our web page contains such credit notice

```
LEOBOT©2018 Created by SYSTEMDI 
Developed by Yaryna Demkiv
```

I suggest to update it to include everyone who works on the project :) Here's how I see it

```
LEOBOT©2018 Created by ROS Ukraine Community
```
where [ROS Ukraine Community]() links to one of our sites, such as

1. Page with list of LeoBot contributors https://github.com/ros-ukraine/leobot/graphs/contributors
2. ROS Ukraine GitHub page https://github.com/ros-ukraine
3. Facebook page https://www.facebook.com/groups/279366215828155

YarynaD, 2018-02-08:
@MaxZhaloba  great idea.

LyubomyrD, 2018-02-08:
@MaxZhaloba +1

AndriyPt, 2018-02-09:
@MaxZhaloba I guess link it to GitHub https://github.com/ros-ukraine . Maybe later we will come up with official web site :)

SystemDiagnosticss, 2018-02-09:
@MaxZhaloba +1

MaxZhaloba, 2018-02-09:
HTML updated in branch F#16_web_server

closed 2018-02-09 by MaxZhaloba