ros-ukraine/leobot #50: Add keyboard teleoperation

AndriyPt, 2018-02-12:
Use this as an example https://github.com/AndriyPt/ros-navigation-examples/blob/kinetic-devel/moon_control/launch/teleop.launch

MaxZhaloba, 2018-02-27:
@AndriyPt, I apparently need your help with this issue.

I'm trying to configure this feature in the branch #F50_keyboard_teleop. I'm performing the following:

1. Start the simulation.launch
2. Start the gmapping.launch
3. Keep the command window on top and press the keys around "K" letter.

I expect the robot to move in gazebo simulation, but nothing happens. Can you please advice what I'm doing incorrectly to reach the goal?

AndriyPt, 2018-02-28:
@MaxZhaloba for teleoperation you only need `Start the simulation.launch` and launch teleoperation node like here https://github.com/AndriyPt/ros-navigation-examples/blob/kinetic-devel/moon_launch/launch/teleop.launch. GMapping is needed for map building for navigation. 
Please have a look on http://wiki.ros.org/twist_mux node and collaborate with @SystemDiagnosticss since he is currently working on joystick teleoperation. I will not be available till Saturday's evening :)

PR #65 (2018-03-23): Add keyboard teleoperation
Closes #50 

@AndriyPt, when I manually start the simulation.launch and control.launch I'm able to control the robot from keyboard, but it works not very smoothly and responsively. Please review my code and let me know if any improvements are necessary.
closed 2018-03-23 by AndriyPt