ros-ukraine/leobot #84: Integrate Navigation widget to web interface

AndriyPt, 2018-03-27:
Add NAV2DJS to right hand side panel on web interface.
It might be used for goal setting or might be just for visualization of current robot position.
http://robotwebtools.org/tools.html#widgets

MaxZhaloba, 2018-03-30:
I'm currently trying to pull the nav2djs module from github via npm. When I configure it in package.json, I receive the following prompt:
```
The authenticity of host 'github.com (192.30.253.112)' can't be established.
RSA key fingerprint is SHA256:nThbg6kXUpJWGl7E1IGOCspRomTxdCARLviKw6E5SY8.
Are you sure you want to continue connecting (yes/no)? 
```
So, even if real user accepts it (which would make it possible to perform a man-in-the-middle attack), the CI servers will fail on timeout, so we cannot use this method. I'm looking into possible solutions.

MaxZhaloba, 2018-03-30:
When I switched to https, this issue went away, but then I got
```
npm ERR! addLocal Could not install /tmp/npm-26041-cbe729c1/git-cache-b960861e/ef7bebf8bdb1919d947afe46ab4b2fb4278039b3
```
This is caused by the fact that npm cannot `install` repositories which contain no `package.json`.

MaxZhaloba, 2018-03-30:
At the moment, all libraries are configured and loaded into browser without console errors.

MaxZhaloba, 2018-04-06:
Documentation on ros.org:
* http://wiki.ros.org/nav2djs/
* http://wiki.ros.org/nav2djs/Tutorials/CreatingABasicNav2DWidget

MaxZhaloba, 2018-04-08:
I've found that we have a TF topic at `/robot_state_publisher/tf` but configuring this value doesn't seem to have any impact.

LyubomyrD, 2018-04-08:
@MaxZhaloba, 
Please use map instead of /map_metadata
You can find all other topic using the following command
rostopic list

I believe that robot pose should be in geometry_msgs, but I can't verify this right now.

MaxZhaloba, 2018-04-09:
Thanks, Lyubomyr. I've just had a second look and it seems that `/map` is a default value for `topic`. I tried several settings but with no luck so far.

@AndriyPt, can you please advise on this?

MaxZhaloba, 2018-04-09:
The docs specify that it's necessary to initialise the `robot_pose_publisher` node. Trying this now.

MaxZhaloba, 2018-04-09:
I've initialised the `robot_pose_publisher` in `web_server` package and pushed the code to git.

MaxZhaloba, 2018-04-09:
Aha, and we need a new instance of `map_server`.

MaxZhaloba, 2018-04-09:
FWIW, running the simulation in default world, gmapping, web server and PyCharm together requires more than 3.6 Gb of RAM or a swap partition. Otherwise, the OS freezes.

MaxZhaloba, 2018-04-09:
In the end, this stack yet requires more than 4 GB of real RAM. Swapping takes too much time, and freezes the interface.

AndriyPt, 2018-04-10:
@MaxZhaloba for comfort development with simulation I would recommend at least 8GB (or even 16GB) of RAM. Regarding sequence of staring things:
1. Launch simulation
2. Launch navigation
3. Launch web_servers

Using rostopic list you should be able to find all needed topics which should be self explanatory.

MaxZhaloba, 2018-04-10:
@AndriyPt, thanks for your help. I have managed to launch this configuration, however any additional program would overflow the RAM.

Still, the widget didn't display any objects and remained black. At the moment I don't have any new ideas how to troubleshoot this issue further.

I'm receiving the following output in web server console:
```
[INFO] [1523387851.687793, 91.500000]: Client connected.  1 clients total.
[INFO] [1523387872.787702, 102.450000]: [Client 0] Subscribed to /wheel_diff_drive_controller/cmd_vel
[INFO] [1523387872.844967, 102.460000]: Client connected.  2 clients total.
[INFO] [1523387874.130276, 103.260000]: [Client 0] Subscribed to /head_position_controller/state
[INFO] [1523387874.134174, 103.260000]: [Client 1] Subscribed to /wheel_diff_drive_controller/cmd_vel
[INFO] [1523387874.740766, 103.610000]: [Client 0] Subscribed to /map
[INFO] [1523387874.747857, 103.610000]: [Client 1] Subscribed to /head_position_controller/state
[INFO] [1523387875.316874, 103.850000]: [Client 1] Subscribed to /map
[INFO] [1523387875.583050, 103.900000]: Client disconnected. 1 clients total.
[INFO] [1523387875.816874, 104.060000]: [Client 1] Subscribed to /move_base/status
[INFO] [1523387876.205529, 104.170000]: [Client 1] Subscribed to /move_base/feedback
[INFO] [1523387876.783990, 104.370000]: [Client 1] Subscribed to /move_base/result
[INFO] [1523387877.188288, 104.630000]: [Client 1] Subscribed to /robot_pose
```
So, it looks like the web client (actually 2 of them as I reloaded the page) successfully connects to ROS to read the necessary topics but for some reason the widget doesn't display the map.

AndriyPt, 2018-04-10:
@MaxZhaloba it would be nice if you provide branch name in such requests :) 
Also you need to upgrade your machine else each task will be hard to accomplish. Please note that we haven't yet started implementation of high level behaviour.

MaxZhaloba, 2018-04-10:
@AndriyPt, here you are: `F#84_nav_widget `

MaxZhaloba, 2018-04-11:
I've just run another test over remote control on a PC with 8 Gb of RAM. When I launched the headless simulation (see readme.md), navigation, web server and browser it took 3 Gb of RAM. As soon as I opened the main page, it allocated another 3 Gb of memory and the PC has frozen. I repeated this twice with Firefox and Chrome.

@SystemDiagnosticss, can you please try to run this configuration when you finish working on the joystick?

MaxZhaloba, 2018-04-11:
Ok, it also freezes in windowed simulation mode on this computer. I've just tried it once again.

AndriyPt, 2018-04-15:
@MaxZhaloba I have done small investigation to run the following example page https://github.com/ros-ukraine/leobot/blob/F_sample_nav_widget/leobot_web_server/web_content/sample.html on my machine.
Steps:

1. Launched NVidia-Docker of leobot
2. Launched Firefox on host machine
3. `roslaunch leobot_launch simulation.launch world_file:=artgallery headless:=True gui:=False`
4. `roslaunch leobot_launch navigation.launch map_file:=artgallery`
5. `roslaunch leobot_launch web_server.launch`
6. Open in Firefox on my host machine address **http://localhost:8080/sample.html**
7. Saw the following after 1-2 minutes of waiting
[image]

My machine used 12 GB of RAM at peak period and 8 GB after map appeared in the map.
So you definitely need more RAM or tune component to have lower refresh rate or implement your own component.
A lot of memory at peak point was consumed by ROS WebSocketServer.

MaxZhaloba, 2018-04-15:
@AndriyPt, thanks for your help. I’ll try to give it more time. Maybe it will fit into 8 Gb of RAM and swap.

MaxZhaloba, 2018-04-16:
Cool. I've just opened the `sample.html` page and gave it several minutes to intialise. The interface was almost frozen, but after some time it displayed the map. Now working on branch #84.

MaxZhaloba, 2018-04-18:
Just an update: I've initialised the first draft version of navigation widget on `index.html`. I'm continuing to investigate why it didn't work at the beginning.

MaxZhaloba, 2018-04-18:
Ok, it turned out that I just didn't wait enough to let it initialize the map.

Here's the current progress on this task:

[image]

And current issues:

1. The map is too small
2. There's no robot on the map
3. The map is rendened only once and not updated when the robot is moving
4. The map is not rotating along with the robot
5. Functionality to set the navigation goal

@AndriyPt, which of the abovementioned issues we should consider important and implement in this task?

AndriyPt, 2018-04-25:
@MaxZhaloba maybe we should implement it ourselves? How long do you think it will take you to download the map from ROS topic and scale it on HTML 5 canvas?

MaxZhaloba, 2018-04-25:
@AndriyPt, that's possible. I'll try to estimate the scope of work. But I'm not sure that the slowdown happens on frontend. It could be on backend. I'd like to check it before we proceed with new widget.

I'm thinking how to measure the values of CPU and RAM usage if we run the robot separately on 2 physical machines - server and client. Do you have any idea where we can get a second computer to benchmark this? Perhaps @SystemDiagnosticss could help?

ENGINEERDK, 2020-01-15:
@MaxZhaloba I suggest you edit your map's PGM file. these files are editable. The hight RAM requirement is due to the loading of the unnecessarily large map.

I am working on ROS2DJS WEB control, any lead???