ros-ukraine/leobot #89: Prevent screen locking on timeout when web pages are open

MaxZhaloba, 2018-03-29:
This issue should be fixed on all pages: main.html and vr.html

In theory, when there's a playing video on the page, browser prevents screen from locking. However we do have video streams on our pages, the screen locking still currently takes place in our case.

Course of action:
1. First of all, we need to investigate how to prevent browser from screen locking using the video streams we have and why this currently does not happen. This is the recommended solution.
2. Only in case if we reasonably would not be able to use the first solution, we can use a third party library for this purpose.

MaxZhaloba, 2018-03-30:
Currently the main page uses `<iframe>` to display the video stream. @YarynaD, is it OK to update this to `<video>`?

YarynaD, 2018-03-30:
@MaxZhaloba there's not much I know about video streaming. So, please correct me if I'm wrong. From documentation 
http://wiki.ros.org/roslibjs/Tutorials/Publishing%20video%20and%20IMU%20data%20with%20roslibjs#CA-3fc6413f79f1645f69db70dc902a3a12ecc54f7c_155
 it seems to me that video is streamed via "jpeg" files.  In this case, <img> tag should be suitable.

MaxZhaloba, 2018-04-18:
Just to mention, the imge from virtual camera is transferred in MJPEG format. More info on this [method](https://en.wikipedia.org/wiki/Motion_JPEG#M-JPEG_over_HTTP).

I've tried to render it using the `<img>` tag and it works, but no luck with `<video>` so far.

MaxZhaloba, 2018-04-18:
I've found out that browser always treats the MJPEG stream as image. So we cannot render it as video in playing state. At the very most I was able to specify the MJPEG URL in `placeholder` attribute of `<video>` but in this case the screen was still locked in iOS Chrome.

@SystemDiagnosticss, I'm passing this task over to you. Feel free to proceed with the approach that uses a plugin for this purpose.

P.S. The method suggested in task #85 is another option to render an MJPEG stream.

SystemDiagnosticss, 2018-04-18:
@MaxZhaloba thanks. I tested NoSleepjs lib in my lockal branch. Its work fine. But now I add its directly in js folder without cmake. This lib need webpack build.

MaxZhaloba, 2018-04-25:
@SystemDiagnosticss, which branch could I use to configure the automatic assembly of NoSleepJS?

MaxZhaloba, 2018-04-26:
@SystemDiagnosticss, I'm thinking that most likely we won't need the NoSleep JS in next release since we'll probably be rendering a real video stream using `<video>` tag and VR API (task #104).