ros-ukraine/leobot #90: Hide the navigation and address bars (go fullscreen)

MaxZhaloba, 2018-03-29:
The web page should occupy full screen size in Chrome, Safari and preferably in Firefox (mobile versions).

iOS Safari and Firefox show 2 additional elements by default: address bar at the top and navigation bar at page bottom.

closed 2018-04-27 by AndriyPt

MaxZhaloba, 2018-04-27:
@AndriyPt, Is this feature no longer needed in Leobot 2 or are you closing it in favour to issue #104?

AndriyPt, 2018-05-02:
@MaxZhaloba from what I have seen on demo it was working. So I have closed it.

MaxZhaloba, 2018-05-02:
@AndriyPt, this might or might not work for us, but just to clarify, the current solution has such limitations:
* Doesn't work on iOS
* Requires a tap to go fullscreen. There's a chance that we might be able to skip it if we implement the approach with VR API (#104).

AndriyPt, 2018-05-02:
@MaxZhaloba so it is working with limitation which is fine to keep it closed. For new VR application we might need something else. So this issue is closed.