ros-ukraine/leobot #96: Robot oscillates in simulation

MaxZhaloba, 2018-04-02:
Steps to reproduce:

1. Run `simulation.launch`
2. Run `teleop.launch` from `leobot_control`
3. Operate the robot and turn left or right
4. Robot starts swaying left to right with about 1 Hz frequency

LyubomyrD, 2018-04-02:
Yes, I know that. My lucky guess is that this is due to PID parameters of the controller especially Kd which deals with oscillation. I tried to tune it but did not give it too much effort.

https://github.com/ros-ukraine/leobot/blob/kinetic-devel/leobot_control/config/gazebo/leobot_control.yml#L17#L19

MaxZhaloba, 2018-04-25:
I've just gave it a try, launching simulation with keyboard teleoperation. I've noticed such changes comparing to `kinetic-devel`
* No oscillation at all
* Reduced speed and better robot control when using keys. In old version the robot was advancing or rotating too fast when I just pressed a key once. Now it starts to move slowly and gains speed if I continue holding the key.

@LyubomyrD, just curious, what was the underlying reason of oscillation?

ldemk, 2018-04-25:
@MaxZhaloba The speed and acceleration limits were inapropriate.

I am glad that robot behaves better now.

ldemk, 2018-04-25:
@MaxZhaloba 
@AndriyPt 
Do You think it is worth approwing the PR ?

LyubomyrD, 2018-04-25:
@MaxZhaloba The speed and acceleration limits were inappropriate

I am glad that robot behaves better now.

LyubomyrD, 2018-04-25:
@MaxZhaloba 
@AndriyPt 

Do you think that it is worth aproving the PR?

closed 2018-04-26 by AndriyPt