ros-ukraine/leobot #108: nvidia-docker won't start

MaxZhaloba, 2018-04-25:
I'm trying to utilize the Nvidia graphic card on my old PC for the sake of experiment and learning, and to measure the FPS rate in Gazebo. I'm using the embedded graphic card "NVIDIA GeForce 8100 / nForce 720a".

I've installed the legacy Nvidia diver using Ubuntu GUI tool "Drivers". Then I've tested that nouveau drivers are no longer used:
```
max@Max:~$ lsmod | grep nvidia
nvidia              11407360  64
drm                   360448  2 nvidia
```

Then I've installed `nvidia-docker`.

In my case `nvidia-modprobe` was not present on my system, so I installed it using apt-get.
```
max@Max:~$ nvidia-modprobe -v
nvidia-modprobe:  version 361.28  (buildd@debian)  
```
After rebooting the OS, when I'm trying to build the docker container I'm receiving the following eror
```
max@Max:~$ nvidia-docker run -it --name leobot_dev -p 8080:8080 -p 8090:8090 -p 9090:9090 -e DISPLAY -e LOCAL_USER_ID=$(id -u) -v /tmp/.X11-unix:/tmp/.X11-unix:rw rosukraine/leobot-dev-nvidia:latest
nvidia-docker | 2018/04/25 14:02:46 Error: Could not load UVM kernel module. Is nvidia-modprobe installed?
```
Please advice what could be the reason of this issue.

AndriyPt, 2018-04-25:
@MaxZhaloba although error is different but this advice helped in my case https://gist.github.com/wangruohui/df039f0dc434d6486f5d4d098aa52d07#common-errors-and-solutions

MaxZhaloba, 2018-04-28:
Thanks. I've tried to install the official binary drivers (legacy version 340.106 in case of my graphic card) as suggested in the article. For some reason it failed and I restored the original nouveau driver, then switched to driver v304 using the "Drivers" application. At some point I wanted to downgrade and try an older official binary driver and then came over the requirements of nvidia-docker, which says:
```
NVIDIA drivers >= 340.29 with binary nvidia-modprobe
```
And my driver version was 340.106 which is probably the reason it doesn't contain the `nvidia-modprobe`. So I just need a newer graphic card which will be supported by the non-legacy driver.

closed 2018-04-28 by MaxZhaloba

AndriyPt, 2018-05-02:
I am using 340.104 and it works for me. I guess  340.106 is >= 340.29

MaxZhaloba, 2018-05-03:
@AndriyPt, is it a legacy driver or an ordinary driver? Was there `nvidia-modprobe` included into the driver or did you install it separately using `apt-get`?
This might be a reason. Because my graphic card is supported (according to Nvidia specs) only by legacy driver and that's what I've installed. `nvidia-modprobe` was not included.