ros-ukraine/leobot #110: Create URDF for Leobot 2.0

AndriyPt, 2018-04-26:


LyubomyrD, 2018-04-27:
@MaxZhaloba I assigned, as we agreed, to help You with this issue

MaxZhaloba, 2018-04-27:
Thanks :) I've already started reading turtorials and watching a video presentation on this.

If you have this info, feel free to add the requirements for desired look of Leobot 2.0 e.g. size and should it use a 4-wheeled mecanum base? I'll begin with something very simple and improve it as I learn.

AndriyPt, 2018-04-27:
Link to Tutorials https://github.com/ros-ukraine/leobot/wiki/FAQ#tutorials-on-urdf

LyubomyrD, 2018-04-28:
@AndriyPt What is the width of aluminum profile?

LyubomyrD, 2018-04-29:
@ros-ukraine/leobot , please, have a look at the CAD model of the Leobot 2.0 equipped with LIDAR and two cameras
[Loebot.PDF](https://github.com/ros-ukraine/leobot/files/1958835/Loebot.PDF)

LyubomyrD, 2018-04-29:
@AndriyPt try to use Adobe Reader

LyubomyrD, 2018-04-29:
This is what I am getting

[image]

LyubomyrD, 2018-04-30:
@AndriyPt please, have a look at the model on GrabCAD https://grabcad.com/library/leobot-2-0-1

MaxZhaloba, 2018-05-01:
@LyubomyrD, I've measured the sizes of the base in PDF file and they appear to be different from the sizes specified by @AndriyPt in drawing above.

Comparison of sizes, in mm:

|Dimension|PDF size|Drawing size|
|-|-|-|
|Base width|460|500|
|Base depth|500|500|
|Base height|260|300|
|Mount height|1430|1450|
|Mount offset|290|290|

Which values should we take?

LyubomyrD, 2018-05-01:
@MaxZhaloba great that You spotted this out! Thank You.
Let us follow the original drawings of @AndriyPt 

Since we do not use the CAD model of the body at this stage, I think we have to focus on more urgent tasks. I will be able to update the size of the Leobot later.

closed 2018-05-22 by AndriyPt