ros-ukraine/leobot #115: Check spherical vision and VR in simulation

AndriyPt, 2018-04-27:
Setup the same approach as here https://www.youtube.com/watch?v=_zMrtydacsQ in simulation.
Try to run code https://github.com/artivis/omnidirectional_rig and check how hard it is adapt it to Google Cardboard VR and Web

LyubomyrD, 2018-05-10:
@AndriyPt @SystemDiagnosticss , please check the PR #122 to verify if a spherical camera is appropriate.

MaxZhaloba, 2018-07-17:
Tutorial on working with OpenCV from ROS:
http://wiki.ros.org/cv_bridge/Tutorials/ConvertingBetweenROSImagesAndOpenCVImagesPython
Posting it here for later use.

MaxZhaloba, 2018-08-02:
FWIW, sample command to visualise the image stream:
```
rosrun image_view image_view image:=/right_fisheye_camera/image_raw
```

MaxZhaloba, 2018-09-02:
Article about restoration of distorted images
https://github.com/abidrahmank/OpenCV2-Python-Tutorials/blob/master/source/py_tutorials/py_calib3d/py_calibration/py_calibration.rst

MaxZhaloba, 2018-09-02:
Another hands-on article on undistortion: https://medium.com/@kennethjiang/calibrate-fisheye-lens-using-opencv-333b05afa0b0