ros-ukraine/leobot #124: Create electronic schematic for robot

nemo8, 2018-05-15:
Schematic will be created in free EDA KiCad.

LyubomyrD, 2018-05-15:
Great choice!

nemo8, 2018-05-15:
lib with STM32F4 board headers: http://smisioto.no-ip.org/elettronica/kicad/kicad-en.htm

nemo8, 2018-05-15:
Добавив 3d модель до футпрінта в KiCad, гарно так виходить :)
[image]

nemo8, 2018-05-21:
[image]

nemo8, 2018-05-21:
KiCad project is located: https://github.com/ros-ukraine/leobot/tree/F%23124_MCU_Board_Schematic/docs/design/hardware/MCU_Board

nemo8, 2018-06-19:
1. Connection diagram:
[image]

2. Preliminary schematic:
[image]

LyubomyrD, 2018-06-19:
@nemo8 great schematics!
It seems like we need both InA and InB for motors drive shield.
I guess that we have to consider powering drive shields from 5V

nemo8, 2018-06-22:
Hi @LyubomyrD ,
according to the motor driver schematic VCC is used only for two signal pull-up (we don't use these pins in our design), so it doesn't really matter 5V or 3.3V:

[image]

connected motor driver signals INA, INB to the stm32 board connector:
[image]

LyubomyrD, 2018-06-22:
@nemo8 as far as I understand, pull-ups are for the motor_enable pin. I guess, that even if we do not use it now, we should add the usage to the schematics for safety reasons and possible power usage optimization (when robot is idle, there is no need for motors to be ebanled). @AndriyPt , please, confirm.

nemo8, 2018-06-22:
@LyubomyrD yes, these pins are for motor enable. For this pins 5V and 3.3V are fine.

Motor is disabled in the next cases 
1. INA and INB = 0
2. INA and INB = 1
3. PMW = 0

[image]

To disable motor using  ENA/DIAGA, ENB/DIAGB  just need to take off supply volatage  (and pininternally will be connected to GND). These pins are dedicated for monitoring fault detection. If we want to disable motor via ENA/DIAGA, ENB/DIAGB pins we need to add the transistor switch for 5V line. 
[image]