ros-ukraine/leobot #126: RPM calculation from encoder value

nemo8, 2018-06-17:
Hi!
Could you please help me with RPM calculation formula? 
Currently, I'm able to read the raw value from the encoder. I've found two formulas to the calculation.

1. https://www.vexforum.com/index.php/16114-calculating-rpms/0
2. https://electronics.stackexchange.com/questions/375971/how-to-calculate-rpm-given-timer-frequency-and-counter-on-stm32

they both look similar, except divider.  Which formula is suitable for us?

Thanks,
Vasyl

AndriyPt, 2018-06-18:
Hey @nemo8 :)
As for me the second links looks reasonable. @LyubomyrD please confirm my thinking :)

LyubomyrD, 2018-06-18:
Hello all.

I guess that the formula moight be the following

Real_RMP_OF_WHEEL = (EncoderCurr - EncoderLast) / sampling_period / CPR / gear_ratio /encoder_correction_multiplier

*encoder_correction_multiplier depends on the setup of the timer. Commonly for stm32f4xx it is either 2 or 4

nemo8, 2018-06-18:
LyubomyrD, thank you!

encoder_correction_multiplier depends on the setup of the timer in my case  = 4.

one motor rotation give me 480 pulses. 

what is CPR ? 

Could you please take paramenters in formula to bracket? (like (((a+b)/4)/8) ).

LyubomyrD, 2018-06-18:
The CPR is counts per revolution - you can find this value in the datasheet of the encoder.

Here is the formula with brackets

Real_RMP_OF_WHEEL = (((((EncoderCurr - EncoderLast) / sampling_period)/ CPR) / gear_ratio) /encoder_correction_multiplier)

nemo8, 2018-06-18:
[image]

LyubomyrD, 2018-06-18:
I guess that You should set the value of the ARR equal to 65535 otherwise the timer might hangup every rotation. Besides, the polarity for the second chanell should be falling, I guess.

nemo8, 2018-06-18:
Now I can read a value of the timer, but I need to investigate more about this timer in the encoder mode...