ros-ukraine/leobot #132: Motor driver discussion

nemo8, 2018-11-30:
Hi,

We have some issue with STM32 board and VNH2SP30 motor driver:
Logic levels for STM32 are 3.0V (3.3V but the board has IC with a diode which generates 3.0 power supply for STM32 ). According to the datasheet of VNH2SP30 (https://www.st.com/resource/en/datasheet/cd00043711.pdf) all input pins are "5 V logic level compatible inputs". (page1), and  "Input high level voltage 3.25" (table 7).  So even we provides 3.0V PWM, this board recognised it as logical 0, and motor will not enabled.

Solutions:
1. Use level shifters like this: 
https://www.ebay.com/itm/3-3V-5V-4-Channel-Logic-Level-Converter-TTL-Bi-Directional-Shifter-Module-10pcs/152613484769?hash=item238878f8e1:g:kN0AAOSwFJBZXdOE:rk:5:pf:0

2. Change the motor driver to other which has "3V-compatible inputs" levels like this:
https://www.pololu.com/product/1451
High-level input voltage  2.1 V (table 7).
https://www.pololu.com/file/0J504/vnh5019.pdf

As for me secon solution is better as we don't need any extra boards.
 

Thanks,
Vasyl

AndriyPt, 2018-12-01:
@LyubomyrD what will you recommend?

nemo8, 2018-12-01:
There is also the second solution, use Opto-isolator IC Module like these:
https://www.ebay.com/itm/TLP281-4-Channel-Opto-isolator-IC-Module-Arduino-High-Low-level-Expansion-Board/191736124728?hash=item2ca45d1538:g:QooAAOSwL7VWlz-R:rk:21:pf:0
https://www.ebay.com/itm/8-Channel-Optocoupler-Module-12V-Isolation-Board-High-Level-Trigger-for-PCB/381752562624?epid=1382231719&hash=item58e239d3c0:g:E9gAAOSwvZZZcMUo:rk:9:pf:0
https://www.ebay.com/itm/TLP281-4-Channel-Opto-isolator-IC-Module-Arduino-High-Low-level-Expansion-Board/191736124728?hash=item2ca45d1538:g:QooAAOSwL7VWlz-R:rk:19:pf:0

and we can leave our red boards (VNH2SP30) as is.

nemo8, 2018-12-01:
As for me, we need to select 1 or 3 solutions. Solution 3 also requires DC/DC isolated module (5v to 5V) but is more reliable than 1.

AndriyPt, 2018-12-01:
So in case of option #3 we need one opto-isolator with 8 channels for whole robot? Why do we need  DC\DC isolated module? (I thought we just need isolated power supply with 5V)
For option #3 is this product good enough https://www.aliexpress.com/item/8-Channel-Optocoupler-Isolation-Board-Module-12V-High-Level-Trigger-Plate-Positive-Phase-Larger-Version/32719949787.html ?

ldemk, 2018-12-02:
I vote for the Opto-isolator IC Module.

@nemo8 can't we get 5V from the pin of Discovery board?

nemo8, 2018-12-02:
@LyubomyrD , @AndriyPt As Lubomyr said we need to protect 4 pins ( InA, InB, EN, PWM). We have 4 drivers, so we need to have  16 channels isolation for the whole robot. It can be one board or separated. 

Isolation is needed for two reasons:
1. For blocking noise from the motor.
2. To protect STM32 and connected PC from 12V if the motor driver is damage. 

We also need 5V supplying the second part of optocouplers. This voltage can' t be the same as we have on STM32, otherwise, isolation will be not working. 

We have two solutions how to get 5V:
1. Use isolation DC/DC (5V to 5V) to get 5V from STM32 board.
2. Use ordinary DC/DC (12V to 5V) and get 5V from the battery or power supply (Motor source)

LyubomyrD, 2018-12-02:
I totally rely on @nemo8 advise with 5V/5V or 12V/5V isolation. If You can provide some theoretical materials to clarify this issue I would be grateful. This is a very common practice in different PCB's and I do not want to make this mistake in the future.

@nemo8  the 12V/5V choice seems more reasonable for me, but what is Your advice?

nemo8, 2018-12-02:
@LyubomyrD ,
https://habr.com/post/386721/

12V to 5V converters without isolation usually are more chipper than isolated 5V to 5V.
We don't need to have big current for 5V, only for pull up(or pull down) up resistors.  It can be LDO like this: https://www.ebay.com/itm/DC6-5-12V-to-DC-5V-1117-5V-600mA-Power-Module-Low-Dropout-Regulator-Anti-reverse/272981540881?hash=item3f8ef7c011:g:wK0AAOSwdx1aMWO8:rk:1:pf:0

AndriyPt, 2018-12-02:
is there any place where we could buy it fast? Also final list is
https://www.ebay.com/itm/TLP281-4-Channel-Opto-isolator-IC-Module-Arduino-High-Low-level-Expansion-Board/191736124728
and 
https://www.ebay.com/itm/DC6-5-12V-to-DC-5V-1117-5V-600mA-Power-Module-Low-Dropout-Regulator-Anti-reverse/272981540881
?

nemo8, 2018-12-03:
@AndriyPt yes, you are right. 

The first item is available in Ukraine:
https://uamper.com/4-%D0%BA%D0%B0%D0%BD%D0%B0%D0%BB%D1%8C%D0%BD%D1%8B%D0%B9-%D0%BC%D0%BE%D0%B4%D1%83%D0%BB%D1%8C-%D1%81-%D0%BE%D0%BF%D1%82%D0%BE%D0%BF%D0%B0%D1%80%D0%BE%D0%B9-%D0%B8%D0%B7%D0%BE%D0%BB%D1%8F%D1%86%D0%B8%D1%8F-%D0%BD%D0%B0%D0%BF%D1%80%D1%8F%D0%B6%D0%B5%D0%BD%D0%B8%D1%8F-12%D0%92-3.3V-5V-TLP281?search=01005

The second can be : https://uamper.com/%D0%A1%D1%82%D0%B0%D0%B1%D0%B8%D0%BB%D0%B8%D0%B7%D0%B0%D1%82%D0%BE%D1%80-%D0%BD%D0%B0%D0%BF%D1%80%D1%8F%D0%B6%D0%B5%D0%BD%D0%B8%D1%8F-L7805CV-LM7805?search=%D1%81%D1%82%D0%B0%D0%B1%D1%96%D0%BB%D1%96%D0%B7%D0%B0%D1%82%D0%BE%D1%80 or simmilar