ros-ukraine/leobot #136: Add position controller for head link

AndriyPt, 2018-12-05:
Add position controller to head link the same way as it was done for version 1.0

MaxZhaloba, 2018-12-21:
Hi @AndriyPt 

I've added the description of head controller to the branch `F#136_head_controller` but I don't know how to test it.

When I switch to the `v1.0` branch, build the project and start simulation, I get such errors:
```
[INFO] [1545413762.385352, 8180.376000]: Controller Spawner: Loaded controllers: head_position_controller, joint_state_controller, wheel_diff_drive_controller
[INFO] [1545413762.392877, 8180.386000]: Started controllers: head_position_controller, joint_state_controller, wheel_diff_drive_controller
/opt/ros/kinetic/lib/gazebo_ros/gzserver: line 41: 18033 Segmentation fault      (core dumped) GAZEBO_MASTER_URI="$desired_master_uri" GAZEBO_MODEL_DATABASE_URI="$desired_model_database_uri" gzserver $final
[gazebo-2] process has died [pid 17991, exit code 139, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver --verbose -e ode /home/user/workspace/leobot/base/src/leobot/leobot_gazebo/worlds/artgallery.world __name:=gazebo __log:=/home/user/.ros/log/ddc0bd5e-0546-11e9-85ce-0242ac110002/gazebo-2.log].
log file: /home/user/.ros/log/ddc0bd5e-0546-11e9-85ce-0242ac110002/gazebo-2*.log
```

Then sending messages to head controller doesn't rotate the head in `v1.0` branch.
```
rostopic pub /head_position_controller/command std_msgs/Float64 0.5
```

Please advice how to approach this issue.

AndriyPt, 2018-12-27:
Hello @MaxZhaloba :)
Why are you switching to v1.0 branch?
In you branch please launch simulation and your controller. List topics using **rostopic list** command and send position command to your controller. Head should rotate. If not please debug the issue.

MaxZhaloba, 2019-01-02:
I haven't found the root cause of segmentation fault issue but after I rebuilt the container this error doesn't occur every time.

I'm using the following command to rotate the camera link:
```
rostopic pub /camera_position_controller/command std_msgs/Float64 0.8
```

closed 2019-02-06 by AndriyPt