ros-ukraine/leobot #145: Simulation fails in kinetic-devel

MaxZhaloba, 2019-02-24:
When I start the simulation in Art Gallery or empty world it fails with the following messages.

This issue occurs specifically after commit c8f37c2e9df4b5774e5e27a7a035a8c832a4850e

```
[INFO] [1551022588.119648, 8180.156000]: Spawn status: SpawnModel: Entity pushed to spawn queue, but spawn service timed out waiting for entity to appear in simulation under the name leobot
[urdf_spawner-4] process has finished cleanly
log file: /home/user/.ros/log/f066e4b6-3849-11e9-87b9-0242ac110002/urdf_spawner-4*.log
[ INFO] [1551022588.573323467, 8180.156000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1551022588.573400953, 8180.156000000]: Starting Laser Plugin (ns = /)
[ INFO] [1551022588.578545128, 8180.156000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[Wrn] [msgs.cc:1808] Conversion of sensor type[multicamera] not suppported.
[ INFO] [1551022589.059815894, 8180.156000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1551022589.060084166, 8180.156000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1551022589.067951783, 8180.156000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1551022589.068994886, 8180.156000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1551022589.212712206, 8180.156000000]: Loading gazebo_ros_control plugin
[ INFO] [1551022589.213050359, 8180.156000000]: Starting gazebo_ros_control plugin in namespace: /
[ INFO] [1551022589.222064854, 8180.156000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [robot_description] on the ROS param server.
[ INFO] [1551022589.405036264, 8180.156000000]: Loaded gazebo_ros_control.
[ INFO] [1551022589.412380339, 8180.156000000]: imu plugin missing <xyzOffset>, defaults to 0s
[ INFO] [1551022589.412436883, 8180.156000000]: imu plugin missing <rpyOffset>, defaults to 0s
[ INFO] [1551022589.488739757, 8180.156000000]: ForceBasedMove using gains: yaw: 1 x: 1 y: 1

[ INFO] [1551022589.488790906, 8180.156000000]: robotBaseFrame for force based move plugin: base_footprint

[ INFO] [1551022589.506967972, 8180.176000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1551022589.519266168, 8180.186000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[INFO] [1551022589.577051, 8180.216000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1551022589.581029, 8180.216000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1551022589.584383, 8180.216000]: Loading controller: joint_state_controller
/opt/ros/kinetic/lib/gazebo_ros/gzserver: line 41:  5474 Segmentation fault      (core dumped) GAZEBO_MASTER_URI="$desired_master_uri" GAZEBO_MODEL_DATABASE_URI="$desired_model_database_uri" gzserver $final
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/controller_manager/spawner", line 207, in <module>
    if __name__ == '__main__': main()
  File "/opt/ros/kinetic/lib/controller_manager/spawner", line 185, in main
    resp = load_controller(name)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 505, in call
    raise ServiceException("unable to connect to service: %s"%e)
rospy.service.ServiceException: unable to connect to service: [Errno 104] Connection reset by peer
[INFO] [1551022589.721128, 8180.216000]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1551022589.721507, 8180.216000]: Stopping all controllers...
[WARN] [1551022589.723185, 8180.216000]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 111] Connection refused
[ INFO] [1551022589.733960424, 8180.206000000]: Physics dynamic reconfigure ready.
[gazebo-2] process has died [pid 5432, exit code 139, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver --verbose -e ode /home/user/workspace/leobot/base/src/leobot/leobot_gazebo/worlds/artgallery.world __name:=gazebo __log:=/home/user/.ros/log/f066e4b6-3849-11e9-87b9-0242ac110002/gazebo-2.log].
log file: /home/user/.ros/log/f066e4b6-3849-11e9-87b9-0242ac110002/gazebo-2*.log
[controller_spawner-6] process has died [pid 5582, exit code 1, cmd /opt/ros/kinetic/lib/controller_manager/spawner joint_state_controller head_position_controller mecanum_velocity_controller __name:=controller_spawner __log:=/home/user/.ros/log/f066e4b6-3849-11e9-87b9-0242ac110002/controller_spawner-6.log].
log file: /home/user/.ros/log/f066e4b6-3849-11e9-87b9-0242ac110002/controller_spawner-6*.log
```

MaxZhaloba, 2019-02-24:
@AndriyPt, does this issue happen at your side too?

AndriyPt, 2019-02-24:
Hey @MaxZhaloba :)
Please pull the latest Docker Image and create new Docker container according to instructions on README.md page.
Rerun commands which you have execute to launch simulation and check if you see such error.
In case if you still see these error please provide exact commands which you are running in this bug report.

MaxZhaloba, 2019-02-24:
Thanks! The simulation works now.

closed 2019-02-24 by MaxZhaloba